digitalpetri/modbus #3: ClosedChannelException

FreedomAlanLiu, 2016-06-17:
An 'ClosedChannelException' occurred while writing, but the 'stateContext' has not changed.

ModbusTcpMaster.java:

             ch.writeAndFlush(new ModbusTcpPayload(txId, (short) unitId, request)).addListener(f -> {
                    if (!f.isSuccess()) {
                        // How to fix? Whether we can?
                        /*Throwable cause = f.cause();
                        if (f.cause() instanceof ClosedChannelException) {
                            stateContext.handleEvent(ConnectionEvent.ChannelClosed);
                            cause = new RuntimeException("Channel closed!", f.cause());
                        }*/
                        PendingRequest<?> p = pendingRequests.remove(txId);
                        if (p != null) {
                            p.promise.completeExceptionally(f.cause());
                            p.timeout.cancel();
                        }
                    }
                });

kevinherron, 2016-06-17:
Hey, I've been meaning to replace the FSM with a pattern I've used successfully in other libraries and I'm pretty sure it takes care of this issue here. Check out the [replace-fsm](https://github.com/digitalpetri/modbus/tree/replace-fsm) branch which has the current implementation.

kevinherron, 2016-12-17:
1.1.0 is released

closed 2016-12-17 by kevinherron