digitalpetri/modbus #4: Modbus client

c-val, 2016-07-11:
I've got a modbus server set up with ip and regular port=502.
I used a [client](http://www.simplymodbus.ca/TCPclient.htm).
Could you give me a hint as to how to implement read features of exposed modbus tables?

thanks a lot,
Charles

kevinherron, 2016-07-11:
I'm not sure if you're talking about just reading some registers, e.g. from the Holding Registers table, which is covered in the README example:

```java
ModbusTcpMasterConfig config = new ModbusTcpMasterConfig.Builder("localhost").build();
ModbusTcpMaster master = new ModbusTcpMaster(config);

CompletableFuture<ReadHoldingRegistersResponse> future =
        master.sendRequest(new ReadHoldingRegistersRequest(0, 10), 0);

future.thenAccept(response -> {
    System.out.println("Response: " + ByteBufUtil.hexDump(response.getRegisters()));

    ReferenceCountUtil.release(response);
});
```

Or if you're talking about something else?

c-val, 2016-07-11:
Well, basically I've got

c-val, 2016-07-11:
a ["DIRIS G"](http://www.socomec.co.uk/range-communication-interfaces_en.html?product=/diris-g_en.html) communication gateways, with all information "made available on the Ethernet network via Modbus TCP".
I've got the IP and the port is regularly set to 502.
I'd like to read the values knowing "slave ID", "Register","register size" and "function code"

kevinherron, 2016-07-11:
Well without some examples I can't really help you, but that's more than enough information to get going with if you understand anything about Modbus. You might want to skim [the specification](http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf) - it's not very lengthy. At the very least you'll come away knowing which function code means which kind of request.

c-val, 2016-07-20:
Hi Kevin, thank you for the spec.
I know that want to issue a request with function code 3 (so ReadHoldingRegistersRequest) to a distant Modbus server, With Register "18442" which I understand is the address in ReadHoldingRegistersRequest's fields.
I must specify a "slave ID". Which I don't find, so I assumed it was master.sendRequest's "unitID" field.
With a Register's length of 2 I get the following :
`master.sendRequest(new ReadHoldingRegistersRequest(18442, 2), 4);`
I get the following response : 0000c346
The thing is, in my ModBus Desktop Client, I have to set "Word Inversion" and "Not Signed" along with "32 bits display".
Does word inversion work with your code?
Thanks

kevinherron, 2016-07-20:
SlaveId and UnitId are the same thing, just different terminology.

Word inversion and interpreting the bytes as unsigned are something your application would choose to do after it go the data from the response. They aren't part of Modbus proper, so not something this library implements. But they're not uncommon.

In this case it's pretty easy since the data is in a `ByteBuf`... something like: `long myValue = response.getRegisters().order(ByteOrder.LITTLE_ENDIAN).readUnsignedInt()`

closed 2016-07-24 by kevinherron