digitalpetri/modbus #8: How to get "DC BA" order from ByteBuf?

vbnetvbnet, 2016-12-10:
How can I get a "DC BA" ordered float from ByteBuf? Like:
[image]

I know very little about modbus. Can you please help me? Thanks.

kevinherron, 2016-12-10:
I'm not sure what that software would consider the order to be by default (maybe DC BA), but try something like this:

```java
short cd = Short.reverseBytes(buffer.readShort());
short ab = Short.reverseBytes(buffer.readShort());
float f = Float.intBitsToFloat(cd << 16 | ab);
```

Assuming the buffer contains at least 4 bytes (meaning you read at least 2 registers, because each register is 16-bit).

vbnetvbnet, 2016-12-11:
@kevinherron Thank you for your reply. I'm trying using your code, but I got the value equal to order BA DC rather than DC BA. Is it because of different default order?

kevinherron, 2016-12-11:
Maybe this then?

```java
short ab = buffer.readShort();
short cd = buffer.readShort();
float f = Float.intBitsToFloat((cd << 16 | ab));
```

vbnetvbnet, 2016-12-12:
It works! Thanks!

closed 2016-12-12 by vbnetvbnet