digitalpetri/modbus #9: Support for modbus serial ?

jvelo, 2016-12-26:
Any interest in supporting modbus serial ?

I could give it a try - at least for the RTU flavor.

Thanks

kevinherron, 2016-12-26:
I've been planning it for a while. Hoping to implement it this week, actually...

jvelo, 2016-12-26:
Wow, that's great :)

If that can be of any help, I'm available for feedback / testing in the field

kevinherron, 2016-12-30:
Hey, still planning on doing this, and yes I'd like your feedback and testing when it's done :)

I'll update this when something is ready. This week hasn't been as "free" as I thought it would be...

awindillman, 2017-01-20:
RTU would be a nice addition.  Would recommend/request purejavacomm (https://github.com/nyholku/purejavacomm) for the serial implementation.  Alternatives (RX/TX or Sun's old one) are not "pure" java and are, at best, outdated.

kevinherron, 2017-01-20:
Yep, still working on it, slowly. Doing it as a version 2.0.0 because it involves major package and module shuffling. I'm planning on using PureJavaComm for the serial implementation.

shubhamkumargupta, 2017-07-07:
Hey, @kevinherron, have you released the support for ModBus Serial? I'm not able too find the RTU support in this repository.

kevinherron, 2017-07-07:
No progress since last comment, but I'll need it for work in the coming months so I suspect that'll force it to get finished.

madisonleavo, 2017-10-05:
@kevinherron How goes serial implementation? do you want to commit to branch if any of us can help?

casmeiron, 2018-03-29:
Hey @kevinherron, I'm too interested on this... ModbusSerial RTU would be gr8! Maybe we can all help you so it can be released sooner. How it goes?

shubhamkumargupta, 2018-03-30:
Hey @casmeiron, we @hashmapinc has developed the Modbus_RTU you can take a look at the forked repo : [](https://github.com/hashmapinc/modbus)

casmeiron, 2018-03-30:
Thanks very much I will take a look into that!

Best Regards.

jaysonjh, 2019-12-17:
Hey @kevinherron, have you released the support for ModBus Serial? I'm not able find the RTU support in this repository？

kevinherron, 2019-12-17:
No, sorry, it never happened. I keep putting it off because the pressure to implement it for use at work went away. I do still plan to do it, but in the meantime you'll have to use some other library if you need RTU/serial support.

kevinherron, 2024-08-23:
Done for 2.0, now available on `master` branch.

closed 2024-08-23 by kevinherron