digitalpetri/modbus #23: Custom FunctionCodes

timbo2k, 2018-12-06:
Hallo 

Is it possible to use modbus library with custom functions codes like 0x41 - 0x48 and 0x64 - 0x6E.
If not yet implemented would it be interesting to add this  feature to library, so that custom manufacturer based functions can also be used with this library? If so I would like to support this great lib.

Best
Tim

kevinherron, 2018-12-06:
This is not possible right now but I will consider it as a requirement for version 2.0, along with serial transport, ASCII protocol support, and the ability to communicate with "modbus-ish" implementations like flow meters that use a non-standard 4 bytes per registers.

timbo2k, 2018-12-06:
Hey Kevin

thanks for your infos. This would really be a nice feature.
If any support is needed I can help with some realization of features.

kevinherron, 2024-08-23:
Should be possible now in 2.0, not proven or tested though.

closed 2024-08-23 by kevinherron