digitalpetri/modbus #24: Atomic operation for single coil requests

simdevmon, 2019-08-16:
I am looking for a modbus library to integrate within an JavaEE application server. 

Basically I need a stateless option to for a single coil request:
```
        ModbusTcpMasterConfig config = new ModbusTcpMasterConfig.Builder(endpoint)
            .setPort(PORT)
            .build();
        ModbusTcpMaster master = new ModbusTcpMaster(config);
        WriteSingleCoilRequest req = new WriteSingleCoilRequest(channel, enabled);
        master.sendRequest(req, 0).thenAccept(response ->
        {
            ReferenceCountUtil.release(response);
        });
```

This method can be called multiple times and even in parallel. I noticed that after this the library still have some open threads. Calling `Modbus.releaseSharedResources();` does not work for parallel or fast consecutive calls. 

I think it would be good if in a method like disconnect all resources are released and provide a thread safe operation.

kevinherron, 2019-08-16:
I've got no problem making `releaseSharedResources()` more thread safe, but it's meant to be called on application or JVM shutdown, not for every request you service.

These resources are relatively heavy and meant to be shared by many instances of `ModbusTcpMaster`.

Further, if your service truly needs to be stateless, then by definition you can't be using these shared resources. Instead, you need to create your own instances of each of these per request and set them on the config builder. This will prevent it from reaching for and possibly instantiating the shared instances.

You'll suffer a performance hit for this. Things like the event loop, and especially the wheel timer, are meant to be shared across an application, not created and stopped constantly.

simdevmon, 2019-08-19:
Thank you for the fast response @kevinherron . I understand your concept better now. My use case is only to write some single coils and I was looking for a lightweight way to do this like open TCP connection, send command, close connection, release resources in a thread safe way.

kevinherron, 2019-08-19:
Try creating your own instances of the shared resources and shutting them down. It may still be performant enough for your usage.

kevinherron, 2019-08-20:
Closing this. I fixed up the synchronization and improved the shared resource stuff a bit for 1.2.0, but you'll still need to configure and manage your own for truly stateless operation.

closed 2019-08-20 by kevinherron