digitalpetri/modbus #30: Slave example exits immediately since resource management changed

mrksngl, 2020-10-09:
In commit 3066f96eb9106643db1d63c1b01d71e784a21edf resource management changed, especially the event scheduler was changed to use a thread factory which creates daemon threads.

This changes the behavior of the library slightly: before the commit, the executor prevented the program from terminating, e.g. after exiting main. After the commit, all modbus threads are daemon threads, so exiting main (more correct: all non-daemon threads) will terminate the program.

You can see this in the slave example, whose only task is to answer the requests of the master (in the background), thus exiting immediately.

I don't know what the best strategy for programs working exactly like the slave example (just responding to requests) is. Except, of course, preventing main to exit by and endless sleep loop, but doesn't feel like "the best" ;)

mrksngl, 2020-10-09:
Btw, JDK and Runtime:

openjdk version "1.8.0_265"
OpenJDK Runtime Environment (build 1.8.0_265-b01)
OpenJDK 64-Bit Server VM (build 25.265-b01, mixed mode)

kevinherron, 2020-10-09:
I've updated the example with a simple `Thread.sleep` call to prevent exiting.

Presumably you're running the slave as part of a larger application with its own lifecycle, but if not then this approach is fine-ish. Certainly good enough for a basic example.

closed 2020-10-09 by kevinherron

mrksngl, 2020-10-09:
Agreed :+1

PR #31 (2020-10-09): Prevent SlaveExample from exiting immediately
fixes #30
commit 0c09cf47 (2020-10-09): Prevent SlaveExample from exiting immediately (#31)

fixes #30