digitalpetri/modbus #36: How to read data in batches

17802592576, 2021-04-06:
I just started using this project, and I'm very confused about how to read data in batches. If you can, please guide me to complete this code.

kevinherron, 2021-04-06:
Batching of data like you mention in your email[1] is a "higher level" operation not implemented by this library.

I'm not familiar with how modbus4j works, but I assume this is some function that takes a list of register address + datatype (mapped over multiple 16-bit registers if necessary) and "optimize" it into a group of requests that would cover all the registers needed.

[1] [image]

17802592576, 2021-04-06:
Thank you very much for your response, I will try other ways to slove this problem.My best to you.

closed 2021-04-06 by 17802592576

iotclouddeveloper, 2024-07-02:
> Thank you very much for your response, I will try other ways to slove this problem.My best to you.

I also have this question，How did you solve it