digitalpetri/modbus #37: Performance problem in 1.2.0

thhart, 2021-05-29:
I have a little problem with the library taking unexplainable pauses from time to time. I have a heavy polling complex application which needs to have reliable poll cycles of 20 ms for each of several devices. The library is working well even with with lower cycles however from time to time there is a noticeable delay for 200-600 ms for all threads using the library. It looks like there is a resource leak going on for this short period and then it is proceeding without problems.
I have traced all outer possibilities without anything strange like JVM pauses, GC pauses, CPU, System I/O, Networking (ping under 1 ms), Slaves itself and so on however I can not exclude anything yet.

I am using the library like this:
```
	public static int sendRead(final ModbusTcpMaster master, ReadDiscreteInputsRequest input, Benchmarker benchmarker) throws Exception {
		if (benchmarker != null) benchmarker.split("REQUEST");
		final CompletableFuture<ReadDiscreteInputsResponse> future = master.sendRequest(input, 0);
		ReadDiscreteInputsResponse response = null;
		if (benchmarker != null) benchmarker.split("RESPONSE");
		try {
			response = future.get();
			return response.getInputStatus().readUnsignedByte();
		} finally {
			if (response != null) {
				ReferenceCountUtil.release(response);
			}
		}
        }
```
Benchmarker is a utility class for tracing the times. The delay is happening in the response section where it is waiting for the result. This method is called up to thousand times per second and it is working but sometimes it starves.
I know it is very hard to say without hands on such a complex system, so my question is rather related to resources sensitive bottlenecks to look for maybe in the underlying Netty world? How can such a problem be traced, are there any tips?

thhart, 2021-05-29:
It turns out it is rather related to some kind of CPU spike during the pauses, so in general keeping CPU at reasonable load seems to be important, however I am wondering if it could be possible to improve this somehow.

thhart, 2021-05-29:
I found this one might be worth to look into, of course this is not the scope of this library but might be used with:
https://netty.io/wiki/thread-affinity.html

kevinherron, 2021-05-29:
You can configure the Bootstrap with `ModbusTcpMasterConfig.Builder::setBootstrapConsumer`.

thhart, 2021-05-30:
I did implement the affinity wrapper however this did not solve it finally. Finally it was an issue in the X710 network card firmware which I updated and all problem did go away. Could not believe it....

closed 2021-05-30 by thhart

kevinherron, 2021-05-31:
Thanks for the update. That’s a wild resolution, never would have figured that out.