digitalpetri/modbus #45: Just help to first use

bergheil, 2022-11-05:
Hi,
I'm using a master simulator, I have created some registries  and I'm trying to read the value using the example documented in the home page of this project.

I'm trying to read a registry to the index 2 using the unitid 1.
Here the code:

```

 CompletableFuture<ReadHoldingRegistersResponse> future =
            master.sendRequest(new ReadHoldingRegistersRequest(2, 3), 1);

        future.whenCompleteAsync((response, ex) -> {
            if (response != null) {
            	//long myValue = response.getRegisters().order(ByteOrder.LITTLE_ENDIAN).readUnsignedInt();
            	//long myValue = response.getRegisters().readUnsignedInt();
            	int myValue = response.getRegisters().readInt();
            	System.out.println("Risultato int = " + myValue);
                ReferenceCountUtil.release(response);
            } else {
                logger.error("Completed exceptionally, message={}", ex.getMessage(), ex);
            }
            scheduler.schedule(() -> sendAndReceive(master), 1, TimeUnit.SECONDS);
        }, Modbus.sharedExecutor());

```

The result is 

`Risultato int = 0`

but the correct value of the registry is "5".
Can anyone help me?

Thanks

Francesco

bergheil, 2022-11-05:
I attach the screenshot of "modpoll" a simple client the read the same registry.
Bye

[image]

kevinherron, 2022-11-05:
Capture the requests from each software using Wireshark and see what’s different.

bergheil, 2022-11-05:
Thanks for the answer, I will do. Can I ask if I'm using in the right way request object?

kevinherron, 2022-11-05:
Yes, but you're reading 3 registers instead of 1 like you are with `modpoll`. 

I'm not sure if `mod poll` uses 1-based or 0-based addressing, either, so the Wireshark capture may help determine the difference.

closed 2023-08-16 by kevinherron