digitalpetri/modbus #49: Unable to use

paulorb, 2023-02-08:
I tried the example code (master) against a Modicom PLC, on latest master "whenCompleteAsync" call of the "sendAndReceive" method is never completed. On wireshark I can see the TCP 3-way handshake completed successfully  but no request at all.

I also tried version v1.1.1, on this one on wireshark TCP handshake ok, on this version I also can see request and response. But the  "whenCompleteAsync" is never called.

kevinherron, 2023-02-08:
Can you post the full code you are using? It sounds like you might just be letting the program exit.

paulorb, 2023-02-09:
Same code you provided MasterExample.java, the only difference is that I ported it to Kotlin and changed the number of masters to 1

`MasterExample(1, 100).start()`

Also I added a delay between requests

```
   for (j in 0 until nRequests) {
                Thread.sleep(1000);
                sendAndReceive(master)
       }
```

On wireshark, each second I see the modbus request and response. But I dont see the message from println

```
 future.whenCompleteAsync({ response: ReadHoldingRegistersResponse?, ex: Throwable ->
            if (response != null) {
                println("response")
                println("Response: " + ByteBufUtil.hexDump(response.registers));
                ReferenceCountUtil.release(response)
            } else {
                println("Completed exceptionally, message=${ex.message}")
            }
            scheduler.schedule({ sendAndReceive(master) }, 1, TimeUnit.SECONDS)
        }, Modbus.sharedExecutor())
```

kevinherron, 2023-02-09:
Have you tried something simple like in the README?

The examples work fine for me, both with the simulated slave and a real PLC, though I had to reduce the concurrent requests down to not overload the PLC.

Try this:
```
ModbusTcpMasterConfig config = new ModbusTcpMasterConfig.Builder("localhost").build();
ModbusTcpMaster master = new ModbusTcpMaster(config);

master.connect();

CompletableFuture<ReadHoldingRegistersResponse> future =
    master.sendRequest(new ReadHoldingRegistersRequest(0, 10), 0);

future.thenAccept(response -> {
    System.out.println("Response: " + ByteBufUtil.hexDump(response.getRegisters()));

    ReferenceCountUtil.release(response);
});

Thread.sleep(Integer.MAX_VALUE);
```

Otherwise you'll have to provide me with a fully self-contained and runnable example, not a partial snippet.

closed 2023-08-16 by kevinherron