digitalpetri/modbus #51: Should I invoke the close() function of the ModbusTcpMaster instance when my collection task finished ?

MengMengDaXiaoJi, 2023-03-20:
My program occurred a terrible **memory leak problem** without invoking the close() function of the ModbusTcpMaster instance.
Should I invoke the close() function of the ModbusTcpMaster instance to solve this problem?

By the way, I found it really use much time to invoke **Modbus.releaseSharedResources()** function compared with no invoking it.

MengMengDaXiaoJi, 2023-03-21:
```
        future.whenCompleteAsync((result, throwable) -> {
            modbusMasterUtil.disposeModbusConnector();
            if (throwable != null) {
                future.completeExceptionally(throwable);
            }
        });
```
This memory leak problem seems to be solved by add this code.

kevinherron, 2023-03-21:
`releaseSharedResources` should only be called when you are done using the library, usually during application shutdown or something like that.

The memory leak could be from responses that did not have their buffers released as the example in the README shows. 

Calling `ModbusTcpMaster::disconnect` should be done if you aren't planning to use that master instance any more, otherwise it may still be "alive" and trying to maintain a connection.

closed 2023-03-22 by kevinherron