digitalpetri/modbus #52: Sending requests simultaneously

kalwinskidawid, 2023-03-22:
Hi,
Can I send multiple inquiries in parallel? 
I have sth like that: 
```java

    private void test(){
    this.readHoldingRegisters(0x1,0x1)->thenCompose(response-> {
    if(response.getRegisters().getUnsignedShort(0) != 255)
                final CompletableFuture<ReadHoldingRegistersResponse> firstPart =
                        this.readHoldingRegisters(firstPartStartAddress - 0x9c40, firstPartEndAddress - firstPartStartAddress);
                final CompletableFuture<ReadHoldingRegistersResponse> secondPart =
                        this.readHoldingRegisters(secondPartStartAddress - 0x9c40, secondPartEndAddress - secondPartStartAddress);

                List<CompletableFuture<ReadHoldingRegistersResponse>> test  =Stream.of(firstPart,secondPart)
                        .collect(Collectors.toList());
                 //...create object etc 
    });
    }
    private CompletableFuture<ReadHoldingRegistersResponse> readHoldingRegisters(final int offset, final int length){
        return this.connection.sendRequest(new ReadHoldingRegistersRequest(offset, length), super.getSlaveId());
    }
```

Usually after `->thenCompose`, only the first part will be called and from it I will get the answer, while from the second part I will not get the answer.  If I comment the first part request then I'll get response from sencond part. But I can't get to the point where both answer in parallel. What's wrong?

kalwinskidawid, 2023-03-22:
Or is it a problem with the simulator? If I do `Thread.sleep(1000)`, between `firstPart` and `secondPart` then everything works fine. I use [ModbusTool](https://github.com/ClassicDIY/ModbusTool)

kevinherron, 2023-03-22:
The library has no limit on sending concurrent requests, however in the real world not all Modbus TCP devices will support it. 

I'm not sure about your simulator, maybe look at a Wireshark capture to see if without the `Thread.sleep` call 2 requests are sent before receiving the response will cause the device to not respond to the first.

kalwinskidawid, 2023-03-27:
Thanks for info, probably I should check it at real Modbus device. Okay then I have to create synchronous requests or `.thenCompose`. What exaclty do `ReferenceCountUtil.release(response)` ? Should I use after every received response or I can skip it?

closed 2023-08-16 by kevinherron