digitalpetri/modbus #67: Usage sample to query a SUN2000 inverter?

StefanOltmann, 2024-06-28:
Hi,

I'm totally new to modbus and I look for a way how I can query my SUN2000 using Java.

Can you provide a sample how I can use this project?

The interface definitions are here:
https://forum.iobroker.net/assets/uploads/files/1683548281975-sun2000ma-v100r001c20-modbus-interface-definitions.pdf

For example I want to get these values:
 - DC Power (32064)
 - Active Power (37113)
 - Charge and discharge power (37001)
 - Battery SOC (37004)
 
How would I do that?

And can I get them all at once or do I need to make separate queries?

Can I get multiple values in one block if I say "give me 37000 and the next 20 values"?

Kind regards,

Stefan

StefanOltmann, 2024-06-30:
Nevermind. I'm going to use j2mod.

closed 2024-06-30 by StefanOltmann

kevinherron, 2024-06-30:
It's not clear from that document, but you either need to read:
- Holding Register 32064
- Input Register 2064

as sometimes Modbus addressing is prefixed with 0, 1, 3, or 4 for Coils, Discrete Inputs, Input Registers, or Holding Registers.

Reading HR 32064 is as simple as the examples show:
```java
CompletableFuture<ReadHoldingRegistersResponse> future =
    master.sendRequest(new ReadHoldingRegistersRequest(32064, 1), 0);
```

Depending on the device, the unit ID may need to be 1 instead of 0.

> Can I get multiple values in one block if I say "give me 37000 and the next 20 values"?

Yes, and this is how it would be possible to read multiple values at once, as long as they reside in the same contiguous block _and_ the device supports it. It's common for Modbus slave/server implementations to return an exception response when a master/client reads an address that has no data defined at it.