digitalpetri/modbus #77: about disconnect

kliein, 2024-09-26:
How can I detect a disconnected connection? I need to perform some business-level operations when the connection is lost, as well as attempt to reconnect.

kevinherron, 2024-09-26:
You can check the status with `ModbusClient::isConnected` or `ModbusClientTransport::isConnected`.

The client will automatically reconnect, with some variation of behavior depending on the values of `connectPersistent` and `reconnectLazy` in the `NettyClientTransportConfig`.

mrksngl, 2024-09-26:
What I do is to use the `CompletableFuture` returned by e.g. the `sendRequest` method...
```
CompletableFuture<> f = modbusMaster.<ReadHoldingRegistersResponse>sendRequest(new ReadHoldingRegistersRequest(…), …)
```
...and then to handle exceptions on the future, using
```
f.whenComplete((answer, t) -> {
  if (t != null && t.getCause() instanceof IOException) { … }
});
```

Looks a bit cumbersome, especially because the exception I'm interested in is a nested exception, but works ;)

The nested exception could also be a `ModbusTimeoutException` or `ModbusResponseException` which you could also be interested in.

kevinherron, 2024-09-26:
My answer was concerning the current 2.0.0-RC3 version of the library.

It seems you're asking about the 1.2.x version? I'm not sure the answer changes much.

kliein, 2024-09-26:
> You can check the status with `ModbusClient::isConnected` or `ModbusClientTransport::isConnected`.
> 
> The client will automatically reconnect, with some variation of behavior depending on the values of `connectPersistent` and `reconnectLazy` in the `NettyClientTransportConfig`.

Understood. I have determined online status based on the isConnected() method and set the connectPersistent property to true, and reconnectLazy to false. Setting reconnectLazy to false means it will automatically reconnect if the connection is lost, right? Instead of relying on manual triggering; what is the purpose of the connectPersistent property?

kliein, 2024-09-26:
> My answer was concerning the current 2.0.0-RC3 version of the library.
> 
> It seems you're asking about the 1.2.x version? I'm not sure the answer changes much.
> 
> I can probably add a listener/callback to `NettyTcpClientTransport` in the 2.x version.

[image]
it`s 2.0.0-RC3

kevinherron, 2024-09-26:
When `connectPersistent` is `true` it means that even if the **initial** call to `connect()` fails it will keep retrying automatically. Otherwise if that initial call fails it will just remain disconnected. 

`reconnectLazy` means the reconnect happens the next time the connection is actually needed, instead of happening eagerly/immediately.

kliein, 2024-09-26:
> When `connectPersistent` is `true` it means that even if the **initial** call to `connect()` fails it will keep retrying automatically. Otherwise if that initial call fails it will just remain disconnected. Only influences the behavior of the initial call to `connect()`.
> 
> `reconnectLazy` means the reconnect happens the next time the connection is actually needed, instead of happening eagerly/immediately.

okay,thanks

kevinherron, 2024-09-26:
I will soon add a connection listener callback as well: https://github.com/digitalpetri/modbus/pull/78

kliein, 2024-09-26:
the issue has been solved