digitalpetri/modbus #82: Split package make it impossible to use 2.0-RC3 with Java 9+

unverbraucht, 2024-11-08:
Hi all,

first of, thanks for this awesome package :)

I have a Java 21 based application that uses modules. I've added both modbus and modbus-tcp maven packages in version 2.0-RC3. When building I get the following error:

Modules modbus and modbus.tcp export package com.digitalpetri.modbus to module org.hibernate.orm.core

This happens when multiple packages export the same package from different Java modules (this is called a split package in Java Modules). I see that the modbus exports com.digitalpetri.modbus with a fair amount of classes but modbus-tcp only contains Netty and ModbusTcpCodec classes in this package (com.digitalpetri.modbus). Would it be possible to move these to a subpackage like com.digitalpetri.modbus.tcp?

If you want to I can also look into building a module-info.class with a modern Java toolchain but retaining Java 8 compatiblity. I think there was a PR around this?

kevinherron, 2024-11-08:
> If you want to I can also look into building a module-info.class with a modern Java toolchain but retaining Java 8 compatiblity. I think there was a PR around this?

No need, version 2.0 has bumped the minimum required JDK to 17.

kevinherron, 2024-11-08:
Oh, looking at your PR I see that I may have misunderstood the issue.

It's not duplicate module names, it's duplicate packages... got it.

kevinherron, 2024-11-08:
I'll release this as `2.0.0-RC4` soon. 

Glad you reported this before `2.0.0` final since it's a breaking API change... that would have been annoying.

kevinherron, 2024-11-08:
Well, I moved the packages, but explicitly specifying `Automatic-Module-Name` in `MANIFEST.MF` has broken the javadoc task for reasons I don't understand yet.

kevinherron, 2024-11-08:
Ok, try `2.0.0-RC4` when it becomes available from central.

closed 2024-11-08 by kevinherron

unverbraucht, 2024-11-13:
> > If you want to I can also look into building a module-info.class with a modern Java toolchain but retaining Java 8 compatiblity. I think there was a PR around this?
> 
> No need, version 2.0 has bumped the minimum required JDK to 17.

That makes things easier. It would still potentially be interesting to have a real java module instead of an automatic module, which allows linking with jlink. I can open a PR if you're interested, but I'd have to merge these changes first.

Regarding -RC4: works perfectly, thanks.

kevinherron, 2024-11-13:
The final `2.0.0` release is available now as well.

I'm not sure I fully understand what the changes you're proposing would be, so a PR would be welcome. As long as it doesn't cause any problems for non-modular projects as well I don't see the harm.

PR #84 (2024-11-08): Use distinct Java packages for each Maven module
- Move `modbus-tcp` classes to `com.digitalpetri.modbus.tcp`
- Move `modbus-serial` classes to `com.digitalpetri.modbus.serial`
- Move `modbus-test` classes to `com.digitalpetri.modbus.test`
- Explicitly assign Java module names rather than relying on automatic module names

fixes #82
commit 30ebb3ee (2024-11-08): Use distinct Java packages for each Maven module (#84)

- Move `modbus-tcp` classes to `com.digitalpetri.modbus.tcp`
- Move `modbus-serial` classes to `com.digitalpetri.modbus.serial`
- Move `modbus-test` classes to `com.digitalpetri.modbus.test`
- Explicitly assign Java module names rather than relying on automatic module names

fixes #82