digitalpetri/modbus #91: Even when TCP connection is dropped

TheWizz, 2024-12-05:
I see there's an isConnected() method on ModbusTcpMaster to poll for connection status. I would prefer some kind of event or notification when the TCP connection is lost, but haven't found any. Am I overlloking something, or is polling isConnected() every now and then the preferred method?

Mike

kevinherron, 2024-12-05:
This was merged, just hasn't been released yet: https://github.com/digitalpetri/modbus/pull/78

TheWizz, 2024-12-05:
Excellent! Any ETA on when this may be released into the wild?

kevinherron, 2024-12-05:
I'll try to release this evening, should have done it a couple weeks ago but I forgot.

kevinherron, 2024-12-06:
@TheWizz [released](https://github.com/digitalpetri/modbus/releases/tag/v2.0.1) and available in Maven Central any minute.

closed 2024-12-06 by kevinherron

TheWizz, 2024-12-06:
Many thanks!